adangert/JoustMania #291: ModuleNotFoundError: No module named '_psmove'

jango-fx, 2023-05-06:
I tried to install JoustMania on a RasberyPi 4 b 8Gb.
Newly installed latest version of Pi OS 
I have run and rerun setup.sh.

The automatice start doesn't happen at all and if i try to manually run the scripts via
```
python ./piparty.py
sudo python ./piparty.py
python3 ./piparty.py
sudo python3 ./piparty.py
./venv/bin/python ./joust.py
sudo ./venv/bin/python ./joust.py
```

it always fails with 
```
ModuleNotFoundError: No module named '_psmove'
```
i believe i used to fail with
```
ModuleNotFoundError: No module named 'psmove'
```
until i manually built psmoveapi. but the `_psmove.pyd` that is supposed to be created does not exist.

now i'm completely lost.
the manually built psmove binary seems to work and at least detect the controllers when connected via usb). the psmoveapi python examples don't work either, though... and i can't determine with certainty if the controllers are paired…

adangert, 2023-05-06:
Hi I've just made some updates to the install script, usually you can start joustmania with the command ./joust.sh inside the JoustMania folder

See if that helps, or try to reclone and reinstall joustmania again

jango-fx, 2023-05-06:
hah! reclone+install did the trick now! 
thx for the quick response. much appreciated 🙏 

now i'm wondering though: can i stop the auto-started game somehow?

closed 2023-05-06 by jango-fx

adangert, 2023-05-06:
Yes sudo ./kill_processes.sh should do the trick, if you look in /etc/supervisor there are all the files for joustmania running automatically on startup. I hope that helps!