adangert/JoustMania #293: Steam Windows version not starting

jango-fx, 2023-05-07:
not sure if this is the right place, but the Steam version doesn't seem to work?

Windows 10 Pro 22H2

that's the console output, otherwise nothing is happening:
```
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
C:\Program Files (x86)\Steam\steamapps\common\JoustMania\pydub\utils.py:165: RuntimeWarning: Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work
win32
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
C:\Program Files (x86)\Steam\steamapps\common\JoustMania\pydub\utils.py:165: RuntimeWarning: Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work
win32
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
C:\Program Files (x86)\Steam\steamapps\common\JoustMania\pydub\utils.py:165: RuntimeWarning: Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
C:\Program Files (x86)\Steam\steamapps\common\JoustMania\pydub\utils.py:165: RuntimeWarning: Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work
win32
 * Serving Flask app "webui" (lazy loading)
 * Environment: production
   WARNING: Do not use the development server in a production environment.
   Use a production WSGI server instead.
 * Debug mode: off
win32
 * Running on http://0.0.0.0:80/ (Press CTRL+C to quit)
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
C:\Program Files (x86)\Steam\steamapps\common\JoustMania\pydub\utils.py:165: RuntimeWarning: Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
C:\Program Files (x86)\Steam\steamapps\common\JoustMania\pydub\utils.py:165: RuntimeWarning: Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work
win32
win32
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
C:\Program Files (x86)\Steam\steamapps\common\JoustMania\pydub\utils.py:165: RuntimeWarning: Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work
C:\Program Files (x86)\Steam\steamapps\common\JoustMania\pydub\utils.py:165: RuntimeWarning: Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work
win32
win32
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
C:\Program Files (x86)\Steam\steamapps\common\JoustMania\pydub\utils.py:165: RuntimeWarning: Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
C:\Program Files (x86)\Steam\steamapps\common\JoustMania\pydub\utils.py:165: RuntimeWarning: Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work
win32
win32
audio/Menu/music\backgroundmusic2.wav
C:\Program Files (x86)\Steam\steamapps\common\JoustMania\scipy\signal\signaltools.py:2223: FutureWarning: Using a non-tuple sequence for multidimensional indexing is deprecated; use `arr[tuple(seq)]` instead of `arr[seq]`. In the future this will be interpreted as an array index, `arr[np.array(seq)]`, which will result either in an error or a different result.
C:\Program Files (x86)\Steam\steamapps\common\JoustMania\scipy\signal\signaltools.py:2225: FutureWarning: Using a non-tuple sequence for multidimensional indexing is deprecated; use `arr[tuple(seq)]` instead of `arr[seq]`. In the future this will be interpreted as an array index, `arr[np.array(seq)]`, which will result either in an error or a different result.
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
C:\Program Files (x86)\Steam\steamapps\common\JoustMania\pydub\utils.py:165: RuntimeWarning: Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work
win32
C:\Program Files (x86)\Steam\steamapps\common\JoustMania\scipy\signal\signaltools.py:2230: FutureWarning: Using a non-tuple sequence for multidimensional indexing is deprecated; use `arr[tuple(seq)]` instead of `arr[seq]`. In the future this will be interpreted as an array index, `arr[np.array(seq)]`, which will result either in an error or a different result.
audio/Menu/music\backgroundmusic2.wav
```

jango-fx, 2023-05-07:
ok. that's happening when plugging in a controller
```
[PSMOVE WARNING] No calibration file found (C:\Users\felix.groll\AppData\Roaming\.psmoveapi\00_06_f7_9c_46_d2.calibration or /etc/psmoveapi\00_06_f7_9c_46_d2.calibration)
[PSMOVE WARNING] Magnetometer in 00:06:f7:9c:46:d2 not yet calibrated.

    Unplug the controller.

    Now press the controller's PS button. The red status LED
    will start blinking. Whenever it goes off, press the
    PS button again. Repeat this until the status LED finally
    remains lit. Press Ctrl+C to cancel anytime.
audio/Menu/music\backgroundmusic2.wav
```

jango-fx, 2023-05-07:
and that's happening when trying to pair:
```
could not write to registry, try running program as administrator
```

jango-fx, 2023-05-07:
manually running `piparty.exe` as administrator (probably impossible from within steam?) doesn't throw an error anymore, but no music and no pairing...

alejost848, 2024-04-04:
I'm having the same issue with the Steam version. I managed to pair a single controller but not any additional ones. I can press the trigger and it outputs `starting game`, but that's it, no audio.

I also get this message:

```
audio/Menu/music\backgroundmusic2.wav
Process Process-4:
Traceback (most recent call last):
  File "multiprocessing\process.py", line 297, in _bootstrap
  File "multiprocessing\process.py", line 99, in run
  File "JoustMania\piaudio.py", line 42, in win_audio_loop
  File "site-packages\pyaudio.py", line 750, in open
  File "site-packages\pyaudio.py", line 441, in __init__
OSError: [Errno -9985] Device unavailable
```

Not sure what to do.