adangert/JoustMania #295: Performance issue on pi3 clone (le potato)

DrDecagon, 2023-06-08:
With the shortage of affordable Raspberry Pi hardware, I've decided to try Le Potato, Libre Computers pi 3B replacement.  This board is compatible with Raspbian, has the same cores with higher clocks, and more ram than the original Pi 3B.  

This game used to work on the Pi 3 but some have mentioned performance issues.
https://github.com/adangert/JoustMania/issues/182
https://github.com/adangert/JoustMania/issues/226

I have gotten the game successfully installed and the controllers paired, but sound does not work (it works fine before installing) and often I cannot connect more than 4 or 5 controllers.   CPU utilization is completely maxed out all the time.  

I have tried with raspian 10 and 11 (and some debian builds) and I have tried some headless images to try to make things as lite as possible.  I've tried altering the setup.sh to account for slightly different repos used on Le Potato.

Can anyone confirm this still works on the Pi 3B or has anyone gotten it to work on Le potato?

DrDecagon, 2023-06-09:
A chunk of the log file
`/home/pi/JoustMania/joust.sh: line 14: /home//JoustMania/venv/bin/python3: No such file or directory
/home/pi/JoustMania/joust.sh: line 14: /home//JoustMania/venv/bin/python3: No such file or directory
/home/pi/JoustMania/joust.sh: line 14: /home//JoustMania/venv/bin/python3: No such file or directory
/home/pi/JoustMania/joust.sh: line 14: /home//JoustMania/venv/bin/python3: No such file or directory
/home/pi/JoustMania/joust.sh: line 14: /home//JoustMania/venv/bin/python3: No such file or directory
/home/pi/JoustMania/joust.sh: line 14: /home//JoustMania/venv/bin/python3: No such file or directory
/usr/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: Your system is neon capable but pygame was not built with support for it. The performance of some of your blits could be adversely affected
  return f(*args, **kwds)
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on all addresses (0.0.0.0)
 * Running on http://127.0.0.1:80
 * Running on http://192.168.1.215:80
Press CTRL+C to quit
logname: no login name
Process Process-4:
Traceback (most recent call last):
  File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/JoustMania/piaudio.py", line 132, in audio_loop
    format=alsaaudio.PCM_FORMAT_S16_LE, periodsize=PERIOD, device='default')
alsaaudio.ALSAAudioError: Device or resource busy [default]
ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred
ERROR:dbus.proxies:Introspect error on :1.5:/org/bluez/hci0: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying
`

adangert, 2023-06-14:
Interesting work! There might be some settings that could improve performance, the period might also work, Joustmania did used to run on a pi3, but the modules have been updated since then too. One way that could also work is to have a setting in the yaml settings that disables the audio changing speeds, thus reducing the load on the cpu.

DrDecagon, 2023-06-14:
Thanks! I've commented out all calls to update.py and all references to performing updates.  It does not appear to be an issue with the update.py conflicting with piparty.  But I'm still not sure I completely understand everything.

I'll try messing with the audio speed change, but cpu seems to be maxed out even in the menu before those speed changes would happen.

DrDecagon, 2023-06-15:
So I've messed with audio a bit, voices are now working while background audio is not.  Still getting 
`alsaaudio.ALSAAudioError: Device or resource busy [default]`
Are the voices and the music given to separate mixers?

Also, now when starting a match and then ending it, I drop to only 80% CPU usage in the main menu and one of the 3 heaviest python3 processes disappears.  Sometimes on game exit, things hang and a python process gets left running, killing it manually throws a piaudio error.  It seems to confirm that the background audio is malfunctioning and is chewing up the CPU.  Maybe preprocessing of audio?

DrDecagon, 2023-06-15:
I have tried running pulseaudio in [system-wide](https://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/SystemWide/) [mode](https://stackoverflow.com/questions/66775654/how-can-i-make-pulseaudio-run-as-root) and now /proc/asound/card0/pcm0p/sub0/hw_params reads
```
MMAP_INTERLEAVED
format: S16_LE
subformat: STD
channels: 2
rate: 44100 (44100/1)
period_size: 44096
buffer_size: 88192
```
This is consistent with what I would expect from pulseaudio.  However, I still see the "Device or resource busy" error.

The even stranger thing is that if I set `device = alsaaudio.PCM(device='null')` in piaudio.py, the busy error goes away as expected HOWEVER, the narrator and sound effects STILL PLAY.  How can this be if I have set the sound device to null?  Are the sound effects and voices using different code (sound mixer) than the background music?

adangert, 2023-06-15:
The sound effects and narrator are using pygame to play them, that's probably why you are seeing that I would think!

DrDecagon, 2023-06-16:
I tried using system libraries instead of venv, with varying results.    I tried feeding `pygame.mixer.init(47000, -16, 2, 4096, devicename='null')` but couldn't get it to accept any device names I threw at it when in the virtual env.  It will accept some names when using system libraries, but it doesn't seem to make changes (system libraries are up to date).  This would be the elegant solution if I could get it to work. 

Eventually I tried just forcing dmix systemwide via /etc/asound.conf.  This works, but narration is a little scratchy.  Now that I understand what I happening, I'll test out some different configs to see what performs best.

ntappin, 2023-07-25:
> Interesting work! There might be some settings that could improve performance, the period might also work, Joustmania did used to run on a pi3, but the modules have been updated since then too. One [...]

Does this mean it will no longer work on the Pi 3? I've been trying to run it on one, and it reboots after install, but nothing happens.

DrDecagon, 2023-07-26:
I'm trying to get this running on AML-S905X-CC which is roughly equivalent to a Pi 3b+.  It works but maxes out the cpu at all times.  Try running the `top` command and see if there are multiple Python processes running near the top of that list.  If so, try connecting a move controller, it should light up white, even if you have no sound.  If that works, I have a workaround if it's only a sound issue.

adangert, 2023-07-26:
I don't think there is anything theoretically stopping it from running on a pi3, I have only tested on pi4s since they launched. @DrDecagon may have the closest solution. The cpu is usually maxed on a pi4 as well.

Also recently the price of pi4s has started to decrease! Hopefully this is a trend

ntappin, 2023-07-26:
I noticed the prices had dropped too, so I picked up a pi4 since it seemed like it would run better there anyway.

I'm able to run the ./setup.sh command and it successfully reboots, however it doesn't reboot into the game, and controllers don't pair.

Is there another command I should be running to get the game to run?

adangert, 2023-07-26:
You can try sudo ./joust.sh to run joust in a terminal and see if there are any issues(this also could be why it's not running on the pi3 at the moment)

ntappin, 2023-07-27:
That did it thanks! Now to fix the sound issue. I've got controllers paired, but no sound yet.

adangert, 2023-07-27:
A simple solution that might work, is making sure that the audio output is through the headphone jack(usually the top right corrner of the screen there is a sound option) and that you have headphones plugged in too.

ntappin, 2023-07-27:
That did the trick! Thank you so much.