adangert/JoustMania #301: Error on pygame.mixer.init

NetSpida, 2023-08-02:
On stock debian on raspberry, I get the following error (even if the setting "play music" is disabled in settings):

```
ALSA lib pcm_direct.c:1206:(snd1_pcm_direct_initialize_slave) requested or auto-format is not available
ALSA lib pcm_dmix.c:1087:(snd_pcm_dmix_open) unable to initialize slave
Traceback (most recent call last):
  File "/home/joustmania/JoustMania/piparty.py", line 1050, in <module>
    InitAudio()
  File "/home/joustmania/JoustMania/piaudio.py", line 274, in InitAudio
    pygame.mixer.init(47000, -16, 2 , 4096)
pygame.error: ALSA: Couldn't open audio device: Invalid argument
```

That crash did not occur again after I installed `python3-pygame`, but I see [it should have been already installed via pip](https://github.com/adangert/JoustMania/blob/b4003256bc341153e046cd3baf8811284f50e8ea/setup.sh#L57).

DrDecagon, 2023-08-03:
I kept running into this issue too, my solution was to revert back to using dmix.  I have been trying to figure it out though, did you install python3-pygame to the system via apt, or to the venv via pip?

NetSpida, 2023-08-04:
I used the debian oldstable package python3-pygame-1.9.6 via apt.

adangert, 2023-08-06:
I just re-installed Joustmania on a fresh install of the 64-bit Raspberry pi OS, and didn't encounter this issue (The game boots up and seems to be running well), it sounds like it's probably related to the version of python3-pygame being installed, as it disappeared after you installed it again?

malcolmrigg, 2024-02-06:
I've started having a very similar issue after an upgrade. I hadn't updated the OS/joustmania in a year or more so decided to try a completely fresh install, but I'm still getting this error. It's not exactly the same as the one above, but close enough that it looks related?

The fresh install was onto a RPi 4, using the 64-bit Raspberry Pi OS Lite. Joust mania install script ran through and completed successfully. On reboot the joustmania process just continually crashes.

Here's the error:

`pygame 2.5.2 (SDL 2.28.3, Python 3.11.2)
Hello from the pygame community. https://www.pygame.org/contribute.html
linux
Traceback (most recent call last):
  File "/home/malcolm/JoustMania/piparty.py", line 1082, in <module>
    InitAudio()
  File "/home/malcolm/JoustMania/piaudio.py", line 274, in InitAudio
    pygame.mixer.init(47000, -16, 2 , 4096)
pygame.error: ALSA: Couldn't open audio device: Unknown error 524`

How did people resolve/get around similar errors?

adangert, 2024-02-06:
Is this on the newest install of Joustmania? I added a commit a few days ago which sets in /etc/around.conf for root to use the hardware device 2, which should be the headphones. 

You can test with sudo aplay <wav file> to see if you are able to play anything

adangert, 2024-02-06:
Also bookworm (the newest version of the pi OS) does not work with pairing controllers, I'm going to try testing with the previous OS bullseye to see if that helps

malcolmrigg, 2024-02-06:
> Is this on the newest install of Joustmania? I added a commit a few days ago which sets in /etc/around.conf for root to use the hardware device 2, which should be the headphones.
> 
> You can test with sudo aplay to see if you are able to play anything

Yes, newest Joustmania and that's the issue - on my Pi 4 the headphones are device 0 (1 and 2 are the two hdmi ports). I changed asound.conf to use device 0 and joustmania started working immediately.

malcolmrigg, 2024-02-06:
> Also bookworm (the newest version of the pi OS) does not work with pairing controllers, I'm going to try testing with the previous OS bullseye to see if that helps

Thanks for the heads up! I'll try it, if it doesn't work I'll revert back to bullseye for now.

Thanks for the quick response :)

adangert, 2024-02-06:
Oh well that's strange! There's probably an automated way to set that. HDMI for mine is 0 and 1