adangert/JoustMania #302: Audio Documentation

NetSpida, 2023-08-02:
The [Hardware section of Readme.md](https://github.com/adangert/JoustMania/blob/b4003256bc341153e046cd3baf8811284f50e8ea/README.md?plain=1#L28) does not mention the requirement for an audio device at all, even though [audio seems to be an integral part of Joustmania](https://github.com/adangert/JoustMania/blob/b4003256bc341153e046cd3baf8811284f50e8ea/README.md?plain=1#L15).

It should be mentioned that an audio device (which kind? how to configure) needs to be present or how audio can be turned of if the environment (to loud?) or player selection (to many different languages) do not allow for audio to be used.

adangert, 2023-08-02:
Hi NetSpida, the audio for the game is processed directly on the pi, and then played back through the headphone jack, if that's not working for you, there may have been a problem with installation

adangert, 2023-08-07:
Updated the Readme to address these issues in commit c04491d and 4050fa6 
Thanks for the input!

closed 2023-08-07 by adangert