adangert/JoustMania #305: pygame.error: mixer not initialized

NetSpida, 2023-08-04:
During startup I get the following error:
```
Traceback (most recent call last):
  File "/home/joustmania/JoustMania/piparty.py", line 1053, in <module>
    InitAudio()
  File "/home/joustmania/JoustMania/piparty.py", line 630, in game_loop
    self.check_game_trigger()
  File "/home/joustmania/JoustMania/piparty.py", line 876, in check_start_game
    self.random_added.append(serial)
  File "/home/joustmania/JoustMania/piaudio.py", line 192, in __init__
    self.sample_ = pygame.mixer.Sound(file=fname)
pygame.error: mixer not initialized
```

NetSpida, 2023-08-04:
Possibly the issue might be caused by my solution to https://github.com/adangert/JoustMania/issues/301

adangert, 2024-02-19:
Fixed for the pi 4 on both bookworm and bullseye from #301

closed 2024-02-19 by adangert