adangert/JoustMania #308: PS4 Move controller not fully pairing.

ekobres, 2023-08-07:
I have several PS3 Move (mini-USB) controllers that pair fine (white globe after connecting to USB.)

I recently acquired some of the newer PS4Move (micro-USB) controllers. When plugged in, after pressing the PS button they flash the power light quickly, then slowly, then nothing. The web UI shows the controller connected. If I power it on after disconnecting USB, one of the BT adapters lights up showing a connection, but the controller doesn’t do anything.

Pairing one of these controllers borks the whole BT setup. After this, none of the previously working PS3 controllers works either. I have to reset BT connections and re-pair them.

Reset button on the controller doesn’t help. I’ve tried this 4 times with the same results.

Are the PS4 controllers expected to work? Is there a trick?

ekobres, 2023-08-07:
Update - turns out this happens when the battery is low.

closed 2023-08-07 by ekobres

adangert, 2023-08-07:
Thanks for the update!