adangert/JoustMania #310: CPU at 100% during game

adangert, 2023-09-09:
Joustmania eats all the CPU processes, we should find a way to not have the controller processes eat up all the available resources. from this comment: https://github.com/adangert/JoustMania/issues/307#issuecomment-1667956765

adangert, 2024-06-16:
Instead of polling all the time in a while loop, we may be able to just look at the buffer for each controller and pull packets as necessary, when we do the rework here: https://github.com/adangert/JoustMania/issues/289

ekobres, 2024-10-28:
The CPU thrashing is caused by a few uncontrolled loops running full throttle.
I have a fork with fixes for these - essentially some strategically placed time.sleep() calls.
To help with  figuring out which processes were thrashing, I added titles to all of the processes.

I used different sleep values for different loops depending on what they were doing - either 10ms, 30ms, or 500ms.

Here's what htop looks like with my changes:

[image]

If you are interested, I will create a pull request.

I have not tested it with Windows - only on a RP 4. There is a lot of concurrency in this project, so it's possible adding these sleeps could expose a race condition or other issue, but so far these changes seem to be working well for me.

janlugt, 2024-10-28:
Amazing! That would come along way to making Joustmania run better on a battery. Right now it really eats through them very fast.