adangert/JoustMania #311: Auto - exclude charging controllers

ekobres, 2023-10-10:
Controllers placed on a charger still behave as available controllers. It would be nice if those controllers were automatically excluded from needing to press a trigger to start a game.

This sort of works if someone uses admin mode to force start a game, but it frequently happens in cases where nobody who knows how to do that is playing the game. 

For a party or con, it’s nice to be able to just pop controllers onto a charger when someone is not using them. 

This feature would make that ability to charge controllers when not being played work perfectly!

adangert, 2023-10-10:
This is a great idea!

adangert, 2023-11-01:
Updated in the latest commit, tested a few times with FFA, when the controller is on and then plugged in it goes dark, and the game can start with a fewer amount of controllers, when it's unplugged it is automatically re-added to the game.

closed 2023-11-01 by adangert