adangert/JoustMania #313: Joust Mania python install error

jawlessfoil9855, 2024-01-15:
Trying to install Joust on a PI 4 2GB model on a 128GB sd card. I have been getting a error. I will attach the setup.log file

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
[setup.log](https://github.com/adangert/JoustMania/files/13934640/setup.log)

adangert, 2024-01-15:
Thanks! Something may have changed with the installation, I will take a look soon

jawlessfoil9855, 2024-01-22:
So I found the fault at line 46 on setup.sh file since I was able to dig deeper to find the exact break point

adangert, 2024-02-05:
fixed with the following change: sudo apt-get install -y python3-virtualenv on line 46!
pairing still seems to be a bit wonky with the new bookworm repo, i.e. it's asking me for a pairing code which it never did before, will have to investigate further.

closed 2024-02-05 by adangert