adangert/JoustMania #314: Trying to start a game with controller charging throws error

mchirlin, 2024-01-25:
Below is the error. Seems to be due to self.out_moves not being aligned with self.moves.  Following line errors:

game_moves = [move.get_serial() for move in self.moves if **self.out_moves[move.get_serial()]** == Alive.on.value and (self.move_opts[move.get_serial()])[Opts.random_start.value] == Alive.off.value  ]

Traceback (most recent call last):
  File "/home/pi/JoustMania/piparty.py", line 1086, in <module>
    InitAudio()
  File "/home/pi/JoustMania/piparty.py", line 662, in game_loop
    self.check_start_game()
  File "/home/pi/JoustMania/piparty.py", line 917, in check_start_game
    self.start_game(random_mode=True)
  File "/home/pi/JoustMania/piparty.py", line 962, in start_game
    #start with only controllers that have pushed trigger
  File "/home/pi/JoustMania/piparty.py", line 962, in <listcomp>
    #start with only controllers that have pushed trigger
KeyError: 'a0:ab:51:35:da:42'

adangert, 2024-01-25:
Thanks for this! I'll test and fix this soon!

mchirlin, 2024-01-27:
I am making updates on my fork, how best do you like to collaborate on things like this?  The updates I am making initially are:

- Adding in a bunch of commenting so I can better understand what's going on?
- Fixing the issue with controllers being dead/alive when charging from above, this required a bit of refactoring.
- Adding a few more things to the webui showing active controllers and ready (trigger-pressed controllers) controllers
- Adding a "Game over" sound effect after Joust ends and playing the "And the winner is X team". I don't have a good way of recreating these audio files to just say "And the winner is Green" instead of "And the winner is Green Team". How did these get created initially?
- Using logging instead of print(

adangert, 2024-01-28:
Hey! Thanks for helping out! You can make a pull request and I'll test it before merging.

mchirlin, 2024-02-05:
I have really hefty update I've been making over the last week or so to refactor all of the games to reuse code and just generally make everything more consistent, e.g. move to enums instead of magic numbers. I still have a couple games left to do (Tournament and Ninja Bomb)

I'll create a PR so you can take a look.  Let me know your thoughts.

adangert, 2024-02-05:
That sounds great! I'll take a look at it an let you know!

adangert, 2024-03-04:
hi @mchirlin I finished fixing a few of the bugs for Joustmania (PS3 controller pairing/pi 5 issues), and have started to look at the work you've done!
I've been trying to reproduce the same issue not being able to start the game with a controller plugged in on the master branch. Currently I've tried plugging in one directly to the pi, that controller light goes out, then pressed the trigger button on all of the rest of the remaining controllers, the game (FFA) seems to start, and it prints out in the console:

we are removing controller: 00:06:f7:8f:fa:3c
we are removing controller: 00:06:f7:8f:fa:3c

which seems to be the Mac address of the controller plugged in, and the game seems to start without issues, I can't seem to trigger the same issue on my end. Is there a specific way that you plugged in a controller (to the pi itself/a different charging solution), or something else I might have missed?

Just curious and trying to understand the root cause of the issue. Thanks for putting in the time for this!

mchirlin, 2024-03-05:
There were a couple of issues that I resolved. All are referenced with FIX
in a comment.
It was a bit ago, but what I remember was:
1. if all moves were charging and you tried to start a game it would just
say over and over, Not enough moves.
2. Sometimes moves were both alive and charging at the same time and which
threw errors when starting a game, don't remember what I did exactly to
make it happen, but I have seem them connect by both usb and
bluetooth simultaneously.

-- 
Michael Chirlin, Appian Senior Architect
Appian Corporation | Level 9, 7950 Jones Branch Dr, McLean, VA 22102
240 888 3845 | ***@***.*** |
<https://twitter.com/appianenabled> |

-- 
 <https://www.appianworld.com/>

adangert, 2024-03-10:
yeah I seem to have been able to replicate the issue, 
Without syncing a controller to joustmania and plugging it in via USB (so that it shouldn't be apart of the game)
Then trying to start a game, triggers the error.
I think there was some problem that when a controller was plugged in, it recognised it in some way as apart of Joust.
Also verified this was fixed in the PR! I'm checking out the rest of the code now

adangert, 2024-03-11:
Fixed from https://github.com/adangert/JoustMania/pull/317

closed 2024-03-11 by adangert