adangert/JoustMania #331: enable_ap.sh not working

leed71, 2024-07-29:
Running ` enable_ap.sh` fails

Errors are
```
mv: cannot stat '/etc/hostapd/hostapd.conf': No such file or directory
mv: cannot stat '/etc/dhcpcd.conf': No such file or directory
chown: invalid group: ‘:pi’
```

and the status of `rc-local` on reboot is

```
# systemctl status rc-local
× rc-local.service - /etc/rc.local Compatibility
     Loaded: loaded (/lib/systemd/system/rc-local.service; enabled-runtime; preset: enabled)
    Drop-In: /usr/lib/systemd/system/rc-local.service.d
             └─debian.conf
     Active: failed (Result: exit-code) since Mon 2024-07-29 17:21:06 BST; 10min ago
       Docs: man:systemd-rc-local-generator(8)
    Process: 815 ExecStart=/etc/rc.local start (code=exited, status=2)
        CPU: 60ms

Jul 29 17:21:05 joust systemd[1]: Starting rc-local.service - /etc/rc.local Compatibility...
Jul 29 17:21:05 joust rc.local[815]: My IP address is 192.168.1.207
Jul 29 17:21:05 joust rc.local[826]: wlan0: interface state UNINITIALIZED->ENABLED
Jul 29 17:21:05 joust rc.local[826]: wlan0: AP-ENABLED
Jul 29 17:21:05 joust rc.local[851]: dnsmasq: failed to bind DHCP server socket: Address already in use
Jul 29 17:21:05 joust systemd[1]: rc-local.service: Control process exited, code=exited, status=2/INVALIDARGUMENT
Jul 29 17:21:05 joust dnsmasq[851]: failed to bind DHCP server socket: Address already in use
Jul 29 17:21:05 joust dnsmasq[851]: FAILED to start up
Jul 29 17:21:06 joust systemd[1]: rc-local.service: Failed with result 'exit-code'.
Jul 29 17:21:06 joust systemd[1]: Failed to start rc-local.service - /etc/rc.local Compatibility.
```

I'll have poke now, but just incase someone already has a fix?

system details
```
# uname -a
Linux joust 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
```
```
# cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

adangert, 2024-07-29:
I think this makes sense, awhile back we upgraded the scripts to deal with other usernames than pi, so I bet changing this would help

adangert, 2024-07-29:
Specifically this line here:

https://github.com/adangert/JoustMania/blob/master/enable_ap.sh#L42

adangert, 2024-07-30:
switching to :$(whoami) instead of :pi seems to help. but doesn't fix the issue completely.
looking into other commands to create a hotspot:

https://www.raspberrypi.com/tutorials/host-a-hotel-wifi-hotspot/

namely: sudo nmcli device wifi hotspot ssid <hotspot name> password <hotspot password> ifname wlan0

but will need for this to work without two dongles (as previously), not sure why things are now broken.

adangert, 2024-07-30:
Also some good instructions to try:

https://www.stevemurch.com/setting-up-a-raspberry-pi-for-ad-hoc-networking-tech-note/2022/12

leed71, 2024-07-30:
Hi Aaron
Thanks for your reply, my use case was just to have the AP mode working in a field. The main issue for me was that WLAN0 didn't have a ip addr, I see that `dnsmasq` tries to fix that but it seems that unless it WLAN0 has an IP address, it wont hand one out, so kinda chicken egg problem.

I have this working now by adding a static entry to `interfaces` for WLAN0
(it was news to me that the wifi adaptor can be a client and an AP, everyday is a school day :) )

adangert, 2024-08-11:
Just fixed this in ff232f8d985e6a91d0295cb23d97f284a056eb0e using the new nmcli commands!

was able to clean up a ton of files, and as bonus, no more restarting required when running ./enable_ap !!

closed 2024-08-11 by adangert