adangert/JoustMania #335: Not booting on the last Raspberry OS version

kurainisei, 2024-08-27:
I tried to install JoustMania on a new Raspberry 5. I kept the stock system in, which I believe is right. I updated everything and followed all the instructions, but when I try to start it, it just reboots into the regular Pi desktop. The only thing I can see when it tries to start is that it fails stopping watchdog, but I am not sure that's the problem. Pi is set up on auto login.
What can I do to make it work?
Thanks for all the help you can give me.

adangert, 2024-08-27:
Thanks for the issue, I was able to test joustmania about a week ago and things were working, I'll have to reinstall and try again, something might have been changed.

adangert, 2024-08-27:
You can try running sudo ./joust.sh to see if any errors pop up

kurainisei, 2024-08-29:
so, I tried to run that and this is what I get:
`2024-08-27 14:42:43,312 - INFO - __main__ - Starting piparty
2024-08-27 14:42:43,942 - DEBUG - __main__ - Loading settings
 * Serving Flask app 'webui'
 * Debug mode: off
Address already in use
Port 80 is in use by another program. Either identify and stop that program, or start the server with a different port.
2024-08-27 14:42:43,958 - DEBUG - __main__ - {'color_lock': False, 'color_lock_choices': {2: ['Magenta', 'Green'], 3: ['Orange', 'Turquoise', 'Purple'], 4: ['Yellow', 'Green', 'Blue', 'Purple']}, 'current_game': 'JoustFFA', 'enforce_minimum': True, 'force_all_start': False, 'menu_voice': 'ivy', 'move_can_be_admin': True, 'play_audio': True, 'play_instructions': True, 'random_modes': ['JoustFFA', 'JoustRandomTeams', 'Werewolf', 'Swapper'], 'random_team_size': 4, 'random_teams': True, 'red_on_kill': True, 'sensitivity': 2}
checking out login@@@@@@@@@@@@@@@@@@@
/home/fede/JoustMania
/root
None
running command: logname
fede
returning: fede

running command: sudo runuser -l fede -c 'cd /home/fede/JoustMania;pwd'
/home/fede/JoustMania
returning: /home/fede/JoustMania

running command: sudo runuser -l fede -c 'cd /home/fede/JoustMania;git fetch'
returning: 
running command: sudo runuser -l fede -c 'cd /home/fede/JoustMania;git diff origin/master --name-only --cached'
returning: 
[]
running command: git rev-parse HEAD
5d2961baaf410fc7092e8b98d4d41a5b6e26bffc
returning: 5d2961baaf410fc7092e8b98d4d41a5b6e26bffc

2024-08-27 14:42:44,652 - DEBUG - __main__ - Number of moves: 0
2024-08-27 14:42:44,652 - DEBUG - __main__ - Moves serial: []
2024-08-27 14:42:44,667 - DEBUG - asyncio - Using selector: EpollSelector
`
Seems there is only a problem with a http port? Not sure it could be that. Any ideas?

kurainisei, 2024-08-29:
[image]

this is what I see what I boot, before the system shuts and restarts. Sorry for the screen photo but am not sure how to access that log.

kurainisei, 2024-08-29:
(sorry for the barrage of messages) I also tried to reinstall the os from scratch, and I am still having the same problem.

adangert, 2024-08-29:
Hey thanks for the update, the log looks fine running it without errors, which is strange, if you plug in an audio jack to the pi (or if you have a pi 5 you need an aux converter cable) music should start playing. However it sounds like that isn't necessarily the issue. 

The other thing I can think of off the top of my head is potentially something with the username fede, which it who work for, you could try the 'pi' default username and see if that changes anything.

On my side I've just been released from the hospital, so I'm on the mend a bit, but should be able to try and look at it this weekend.

kurainisei, 2024-08-29:
ooof poor you, sorry to hear that! No worries, I will try the different username and run some more tests, most likely it's my problem. let me see.

kurainisei, 2024-08-29:
Ok, now this is embarrassing. I figured the problem out. My adaptor for the speakers was defective, and so I couldn't hear the sound. It was actually working! I need to try a bit more the pairing and everything else but I think it's there. I am so sorry for this, especially given you've been in the hospital. I got tricked because desktop+no sound+Pi rebooting twice. Gosh. I am closing this. Again, very sorry.

closed 2024-08-29 by kurainisei