adangert/JoustMania #339: How To Guide needed for how to switch bluetooth adapters when new ones are added.

poke50uk, 2024-09-12:
My new bluetooth dongle arrived today. I had already installed using the internal bluetooth for general testing. I seem to only be able to connect 4 or so controllers. I can;t see anywhere any report what bluetooth device JoustMania thinks it sees and is using, and there's no guide to how to swap to ensure that the onboard device is off once installation is complete or if new hardware arrives. It's also not clear with my multiple dongles, how I know which controllers are on which device (i.e. I have 11). 

I'm trying to download various tools to allow me to see which BT adapters are regonised etc, but again - ideally this would be in some tips section somewhere.

adangert, 2024-09-12:
Hi, the best way is to look in /var/lib/Bluetooth there will be a bunch of folders there each representing a Bluetooth dongle, or potentially the internal one if it has not been disabled yet.

You can cd into each folder and see which controllers are connected to each. You can run the reset Bluetooth devices script to remove these and try to pair again. Also deleting the folders and rebooting can also help.

poke50uk, 2024-09-12:
Yes, but just trying to run headless looking via a phone, that's not ideal. Ideally I need to be able to see the state within the Web UI if possible. 

As a linux newbie I couldn't see how to get to the Bluetooth folder as PiOS blocks you, then had to look up how to get tehre in cmd, then had to look up how to get permission to look in there, then it wouldn't regonise my password with SU -, then I stumbled across setting a password which is one that somehow isn't the same as the one you enter when you start the pi up for the first time, then I had to guess to do sudo su - , then I got to the bluetooth folder and it told me 3 different mac adresses, then I had to guess which was which adapter because I should have had 4, then I've had to debug for ages why my new dongle isn't being regonised, and that PiOS is behind on its kernal and Bluetooth 5.0 dongles don't seem to be supported yet but I could try downloading the firmware because even though it was regonised as a usb device it wasn;t being used in some way, then how to do that, found a bluetooth manager, eventually could get there and rename devices and find that it's very rarely showing, and now I can see that the 11 Move controllers, 6 are still pairing internal bluetooth, 1 is pairing to one old dongle, and the other 4 are not connecting to any dongle.

So you can see, not exactly user friendly the whole thing! Need to be confident in finding out a lot of stuff yourself, so adding that into the guide and as much info as you can into teh web UI would help new users out a lot.