adangert/JoustMania #341: Unable to pair more than 7 Move controllers : ionctl (GFEATURE): Broken Pipe

poke50uk, 2024-09-12:
I have 11 Move controllers. I can pair different combinations of them, but only up to 7. They are a mix of very old ones and re-released micro editions for VR. 

I have 
Onboard Bluetooth
Cheapo Bluetooth Dongle A
Cheapo Bluetooth Dongle B
TP Link Bluetooth 5.0 Dongle

It doesn't seem to matter which extra dongles I have in on startup, it seems like all are only ever connecting to the onboard bluetooth. No extra controllers want to connect to any of the other dongles. I have tried one dongle at a time, and changing which are in my hub and which are in the Pi itself. What I get when running from console is a lot of a suspious "ionctl (GFEATURE): Broken Pipe" logging.

adangert, 2024-09-12:
Did you run setup.sh with the

--disable_internal_bt
Command, if so you shouldn't see it in /var/lib/Bluetooth 

If you still do you can delete that folder and reboot the pi.

adangert, 2024-09-12:
Note you definitely do not want to use internal Bluetooth, it's really bad usually. Two dongles should be enough to support 11 controllers. 

Joustmania pairs new controllers to the Bluetooth adapter it finds with the lowest amount of connected controllers. I would wipe out the connections and just try to pair with Bluetooth adapters that you can verify work. Sometimes adapters are broken and dont work well, the best ones are the ones that are linked in the readme.

poke50uk, 2024-09-13:
And I still get a long list of Broken Pipe errors.

poke50uk, 2024-09-13:
Another fresh install. It seems as though if Bluetooth Manager (Bluez) is installed, it stops any new connections. Only combination I've found that works is new PiOS, install Joust Mania straight away without the -internal command, plug in my dongles, restart, connect controllers (still limited to 7), restart, and then those 7 can be played with again on future boots.

Onboard Pi5 at least has bluetooth 5.0.

adangert, 2024-09-13:
it definitely sounds like there is a problem with the dongles themselves, on a fresh install of the PiOS they should be showing up under/var/lib/Bluetooth and controllers should be able to connect to them. And if you disable the internal Bluetooth and none of the controllers are connecting or showing up under any dongles, I would suspect that's the case.

In the past month I've set up a few Joustmania systems using the dongle from the instructions: https://a.co/d/6GG35Um
Without any issues, so it might just be bad hardware, or some sort of Bluetooth hardware that is acting differently than what might be expected. I'm assuming that you are using different dongle types? If so that would be good information to warn others before installing the system.

It sounds like you are able to get things working with the internal Bluetooth, (that's also probably how the 7 controllers are connecting), should work a bit, I would test it out, but that might be the best before your birthday party tomorrow. Sorry you've had to deal with so much frustration through this.