adangert/JoustMania #347: Game crashes after boot

jawlessfoil9855, 2024-11-24:
After a couple fresh installs of PI OS (bookworm) on a raspberry pi 4. After going through the setup.sh and the pi reboots. As soon the pi boots I get an audio of IVY saying Joust updated then it crash. I have attached the log file 
[20241123-23_00_29.log](https://github.com/user-attachments/files/17891986/20241123-23_00_29.log)

adangert, 2024-12-23:
Will need to look into this, 32 bit legacy Pi OS might be the working solution for now to use JM.

adangert, 2025-03-16:
Just tested on the Pi5, things seem to be working well there. I was not able to get things working on the pi 4, I'm going to open a separate issue for now.

closed 2025-03-16 by adangert

adangert, 2025-04-07:
Just created some extra fixes, the game should now work on the pi4, from a clean boot!